fkie/multimaster_fkie #132: Discover ROS-Masters across multiple subnets on one host

uulm-mschoen, 2020-07-17:
I want to synchronize ROS-Cores on a network with three machines, two of them are connected to the third on two separate network interfaces. I created a different subnetwork for both hosts, e.g. 192.168.1.0, 192.168.2.0.
When running the master_discovery I can only connect to one subnetwork at a time using `rosrun fkie_master_discovery master_discovery _interface:=192.168.1.0`. When I don't specify the interface, the hosts cannot find each other. I tried running two master_discovery nodes, one on each local subnetwork, but that doesn't work.
Is it possible to configure the master_discovery_node to listen to both subnetworks? Or is there another possibility to make it work?

atiderko, 2020-07-17:
Discovering: if you run master_discovery on C without `_interface`-parameter it listens on all interfaces but the mutlicast messages for dicovering are send only to default route or route for multicast messages. You can fix the discovery by setting `_robot_hosts=[A,B]` to send unicast discovery messages. (Also on A/B hosts `_robot_hosts=[C]`)

Synchronization: if I understand it correctly you have machine A, B wich are connected to C on two separate network interfaces.
C is not a router => A and B can not communicate => ROS-Topics on A and B can be not connect to eachother.

uulm-mschoen, 2020-07-20:
I got it using `_robot_hosts`. Thank you very much!

closed 2020-07-20 by uulm-mschoen