fkie/multimaster_fkie #133: How to connect two computers

lijian636, 2020-07-27:
How to connect two computers? I always have problems. Can you tell me how to do it

atiderko, 2020-07-27:
did you tried these steps:
http://wiki.ros.org/multimaster_fkie/Tutorials/Setup%20a%20ROS%20master%20synchronization

If so can you tell me which problems you have, please.

lijian636, 2020-07-29:
When I open the interface on one host, I start the discovery node on another host. At this time, when the interface detects the discovery node, the interface will automatically close and the terminal will output the abandoned. I want to know what to do

atiderko, 2020-07-29:
Can you provide me some log outputs (of master_discovery, master_sync,
node_manager) or screenshots, please.

What is an interface? Is it a terminal?

closed 2020-09-18 by atiderko