fkie/multimaster_fkie #134: default_cfg_fkie

SunggooJung, 2020-08-05:
until the 'melodic-devel' branch, executing a launch file from node_manager GUI window was possible by bring up the node like the below.

```
<node name="test" pkg="default_cfg_fkie" type="default_cfg" clear_params="true">
      <param name="capability_group" value="/SYSTEM"/>
      <param name="package" value="capability_scout"/>
      <param name="launch_file" value="test.launch"/>      
 </node>
```

however, for the new version of node_manager this seems now available any more.

Do you have any idea?

atiderko, 2020-08-05:
use `fkie_node_manager_daemon` instead.
```
  <node name="node_manager_daemon" pkg="fkie_node_manager_daemon" type="node_manager_daemon"
      args="-l $(find capability_scout)/launch/test.launch">
  </node>
```
-l: loads given file on start
-a: loads given file on start and launch nodes after load launch file

comma separated for multiple files

SunggooJung, 2020-08-05:
Thank you for your quick advice

my test.launch in capability_scout package is like this,

```
<launch>
  <!-- === GENERAL ARGUMENTS === -->
  <arg name="robot_namespace"          default="scout"/>
  <arg name="simulation"               default="true"/>

  <param name="capability_group" value="test"/>
  <!-- === Sensor On === -->

  <include file="$(find tensorflow_object_detector)/launch/detection.launch">
    <arg name="robot_namespace"  value="$(arg robot_namespace)"/>
  </include>

  <include file="$(find usb_cam)/launch/usb_cam.launch">
    <arg name="robot_namespace"  value="$(arg robot_namespace)"/>
  </include>

  <include file="$(find artifact_localization)/launch/artifact.launch">
    <arg name="robot_namespace"  value="$(arg robot_namespace)"/>
  </include>

</launch>
```
However, I still cannot execute a launch as I have done in 'melodic-devel' branch. Did I correctly configure my test.launch?

atiderko, 2020-08-05:
sorry, my fault!

Instead of `args="-l $(find capability_scout)/launch/test.launch"`
You have to use `args="-l pkg://capability_scout/launch/test.launch"`

SunggooJung, 2020-08-05:
Great! Finally it works now,

Thank you very much!

closed 2020-08-05 by SunggooJung