fkie/multimaster_fkie #135: Issue with rosmsg when running master_discovery

akhilveeraghanta, 2020-08-26:
Steps to reproduce:
`ROS_IP=127.0.0.1`
`ROS_MASTER_URI=http://localhost:11311`
`roscore`
`rosrun master_discovery_fkie master_discovery`
`rosrun master_sync_fkie master_sync`

And this exception happens:
Stack trace:
~~~
Exception in thread Thread-14:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 660, in handle
    requests = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 743, in receive_once
    raise TransportException("receive_once[%s]: DeserializationError %s"%(self.name, str(e)))
TransportException: receive_once[/master_sync/get_sync_info]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
~~~

Ive sourced `/opt/ros/kinetic/setup.bash` and all the other ros nodes work perfectly.
Calling `rosservice call /master_discovery/list_masters` directly also raises the same exception

Also all other service like `/master_sync/get_loggers` work fine, and every other service on other nodes work fine.

atiderko, 2020-08-27:
Hi, I cannot reproduce this error. This error is a bit strange, because I don't use `rosmsg` either in `master_discovery` or `master_sync`.

How did you install multimaster, via debian packages or from source?

Did you try to install **ros-kinetic-rosmsg**?

akhilveeraghanta, 2020-08-27:
Thank you for the fast response, we just upgraded all the packages and it worked again. Must have been some weird dependency mismatch.

closed 2020-08-27 by akhilveeraghanta

atiderko, 2020-08-27:
Thank you for letting me know!