fkie/multimaster_fkie #136: Connection between desktop computer and raspberry

OtaviodaCruz, 2020-09-06:
Hello everyone. I'm using fkie multimaster for 2 machines, a **desktop computer and a raspberry.**

To test whether the communication used the basis of the turtlesim example. Basically I started a turtlesim node in the **rasp**. So, when running `rostopic list` on the **desktop computer** I get the following result:
```
/master_discovery/changes
/master_discovery/linkstats
/rosout
/rosout_agg
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose
```
This seems to suggest that communication is working. The problem is when I try to control the turtle's movements on the **desktop computer**. Using something like:

`rosrun turtlesim turtle_teleop_key`
or
`rostopic pub /turtle1/cmd_vel geometry_msgs/Twist -r 1 -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]'`

When trying to execute these commands to control the turtlesim node created in rasp nothing happens: /
I hoped I could control the turtle. Anyway, my question is: is what I am trying to do possible? Or is it off limits to fkie?

atiderko, 2020-09-07:
Hi OtaviodaCruz,

your scenario is what multimaster_fkie was developed for!
But there are some limitations due to ROS communication establishement between topics. Internally ROS uses hostnames and connects using these names. Please test if names are resolved to the right IP addresses.

Please set also the ROS_MASTER_URI to hostsname, e.g. export ROS_MASTER_URI=http://rasp:11311 and not to localhost, see:
http://wiki.ros.org/ROS/EnvironmentVariables#ROS_MASTER_URI

You can try to start 'node_manager'. Some of the problems could be shown in the GUI.

OtaviodaCruz, 2020-09-07:
Thanks for the answer.
In this case, before I need to configure my etc / host file for the IP and names of my computers, right?
I will try as you indicated. Do you have any explanation about the need to enter a host name? Because in my mind it was supposed to work with localhost too

atiderko, 2020-09-08:
Multiamster FKIE does not need the hostname in ROS_MASTER_URI and in most cases it works also with `localhost`. But if you have some troubles you should set it to the name of localhost, see official explanation: `Great care should be taken when using localhost, as that can lead to unintended behaviors with remotely launched nodes. `

Since ROS uses the hostname to communicate, you need to enter all hosts with ROS into your `/etc/host`.

You can also try to set `ROS_IP`, but it has also limitation since it binds to specified device. If this device is WiFi and you lost the connection the local nodes can't communicate, too.

OtaviodaCruz, 2020-09-12:
Got it, it's more of a precaution.

I will describe my unsuccessful test below.
Well, first I changed the hosts files, for each computer (desktop and raspberry)
[image]

Firt I executed in desktop computer:

`export ROS_MASTER_URI=http://pc1:11311`

And in my rapsberry:

`export ROS_MASTER_URI=http://pc2:11311`

Then, I executed the following commands on each machine
```
roscore
rorun master_discovery_fkie maste_discovery _mcast_group: = 224.0.0.251
rosrun master_sync_fkie master_sync
```
So, in the rasp I do:

`- rosrun turtlesim turtlesim_node`

As expected, a window with a turtle in the middle appears. 

[image]

Next, I make the following command on the desktop computer:

`- rostopic list`

The result is:

```
/master_discovery/changes

/master_discovery/linkstats

/rosout

/rosout_agg

/turtle1/cmd_vel

/turtle1/color_sensor

/turtle1/pose
```
Apparently fkie is working. The problem is when I try to control the turtle's movements on the desktop computer. Using something like:

`rosrun turtlesim turtle_teleop_key`

or

`rostopic pub /turtle1/cmd_vel geometry_msgs/Twist -r 1 -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]'`

But if I execute the command on the rasp (where the turtle node was started) it works perfectly.  
The strangest thing is that when executing the command rqt_graph everything seems to be working perfectly.

[image]

I use melodic, in my desktop computer with ubuntu 18.04 and my rasp ubuntu mate 18.04.

atiderko, 2020-09-12:
That is very strange behavior. If you run
```rostopic pub /turtle1/cmd_vel geometry_msgs/Twist -r 1 -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]'```
and nothing happens it could be a problem to  establish a connection between turtlesim and rostopic. This should be done by ROS itself. Master_sync registers only the connection information in roscore. And that seems to work ;) unless they're wrong...

Are there some wranings in the output of `master_dycovery` or `master_sync`?

Did you checked the info of the /turtlesim node? It should look like:
```
robot@control:~$ rosnode info /turtlesim 
--------------------------------------------------------------------------------
Node [/turtlesim]
Publications: 
 * /turtle1/color_sensor [turtlesim/Color]
 * /turtle1/pose [turtlesim/Pose]

Subscriptions: 
 * /turtle1/cmd_vel [geometry_msgs/Twist]

Services: 
 * /clear
 * /kill
 * /reset
 * /spawn
 * /turtle1/set_pen
 * /turtle1/teleport_absolute
 * /turtle1/teleport_relative

contacting node http://tiderko5:44023/ ...
Pid: 3559
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound (43661 - 192.168.178.104:54332) [27]           <--- local connection
    * transport: TCPROS
 * topic: /turtle1/cmd_vel
    * to: /rostopic_20020_1599895127680 (http://control:35131/)          <--- remote connection
    * direction: inbound (54398 - ros2:42761) [33]
    * transport: TCPROS
```
And your `rqt_graph` looks like (except node_manager, node_manager_daemon): 
[image]

Did you tried `roswtf`?

OtaviodaCruz, 2020-09-13:
There are no errors in the master_dycoveryor master_sync output. Just a few WARN:

[image]

I have now tried to check the node information. Finally I see an error impression

[image]

It prints the information but prints an error "ERROR: Communication with node [http: // otavio-desktop: 46633 /] failed!". I thought it was strange, if he has communication problems, how can he press the information?

When I start the turtle node in RASP, I have

[image]

Again it looks like the communication was successful, since it detected the turtle node.
And when I try to control the turtle's movements on my desktop computer I have

[image]

In this image, it does not appear that the desktop computer is accessing / publishing on the turtlesim node created in RASP.

I'm sorry, I didn't know roswtf, I'm relatively new to ROS. When executing, I got

[image]

OtaviodaCruz, 2020-09-13:
When I run
`rostopic pub / turtle1 / cmd_vel geometry_msgs / Twist -r 1 - '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]'`
I have roswtf
[image]

So sad :/
Communication is really not working.

atiderko, 2020-09-14:
can you ping the `otavio-to-be-filled-by-e-o-m` from your desktop PC?

The warning in `master_discovery` is in your case an error because it can not resolve this name.

OtaviodaCruz, 2020-09-14:
Good news, it worked, before describing what I did I will answer the last message.

Yes, I can

[image]

I realized that even using

`export ROS_MASTER_URI=http://pc1:11311`
and
`export ROS_MASTER_URI=http://pc2:11311`

my roscore still appeared as otavio-desktop (PI) and  otavio-To-be-filled-by-O-E-M (Desktop computer), look

[image]

In addition there is the Warning that you mentioned yourself.
So I replaced pc1 and pc2 of my hosts file with

192.168.100.28  otavio-desktop
192.168.100.5   otavio-To-be-filled-by-O-E-M

After that everything worked perfectly.

I still don't know why it didn't work as pc1 and pc2, there must be some configuration that is making my roscore always run like otavio-desktop and  otavio-To-be-filled-by-O-E-M.

Thanks a lot for the help :)

atiderko, 2020-09-18:
This are good new :-)

_I still don't know why it didn't work as pc1 and pc2, there must be some configuration that is making my roscore always run like otavio-desktop and otavio-To-be-filled-by-O-E-M._
This is default behavior of ROS. You can overwrite it by setting ROS_HOSTNAME e.g `export ROS_HOSTNAME=pc1`
I also did not realized that you used different names for the same host.
But all's well that ends well!

closed 2020-09-18 by atiderko