fkie/multimaster_fkie #138: Unable to start remote node

lijian636, 2020-09-28:
When I start the remote node, the log will remind me of such an error
WARN] 28.09.2020 18:22:22.601289 (/node_manager:master_view_proxy.py:start_node:1921): Error while start '/hdl_graph_slam_nodelet': Error while start node '/hdl_graph_slam_nodelet': Traceback (most recent call last): File "/home/pi/catkin_ws/src/multimaster/fkie_node_manager_daemon/src/fkie_node_manager_daemon/launch_servicer.py", line 526, in StartNode launcher.run_node(startcfg) File "/home/pi/catkin_ws/src/multimaster/fkie_node_manager_daemon/src/fkie_node_manager_daemon/launcher.py", line 240, in run_node if addr in set(ip for _n, ip in get_local_addresses()): File "/home/pi/catkin_ws/src/multimaster/fkie_node_manager_daemon/src/fkie_node_manager_daemon/launcher.py", line 240, in if addr in set(ip for _n, ip in get_local_addresses()): ValueError: too many values to unpack

atiderko, 2020-09-28:
thank you for reporting!

The master branch should be fixed now.

carleft, 2020-09-30:
How can I solve this problem，could you please tell me?

atiderko, 2020-09-30:
I applied the fix to the master branch.
Please update, checkout the **master** branch and build again. Also on remote host!

If the error still occurs provide more info, please, e.g.  ROS_MASTER_URI, ROS_HOSTNAME or ROS_IP if set, hostname.

lijian636, 2020-09-30:
> I applied the fix to the master branch.
> Please update, checkout the **master** branch and build again. Also on remote host!
> 
> If the error still occurs provide more info, please, e.g. ROS_MASTER_URI, ROS_HOSTNAME or ROS_IP if set, hostname.

Sorry, I did another test. I used fkie  to start the node on my local computer, the same error will occur. Is it because my local configuration is too low

atiderko, 2020-12-08:
please reopen, if the problem still occurs

closed 2020-12-08 by atiderko