fkie/multimaster_fkie #139: Network problems

lijian636, 2020-10-21:
Excuse me, I recently encountered a problem in the multi host network connection. Can fkie/ multimaster_fkie be used under the local area network without connecting to the public network.

roehling, 2020-10-21:
Yes

lijian636, 2020-10-22:
When I connect to the local area network and not connect to the public network, the following error will appear when I run fkie node_manager. I can't even read the local launch file. The terminal will prompt the following warning. If you know what's going on, can you tell me? Thank you very much

[WARN] [1603362611.331144]: Error while get_version from grpc://lj-lenovo-xiaoxin-700-15isk:12321: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "Deadline Exceeded"
	debug_error_string = "{"created":"@1603362611.326298968","description":"Deadline Exceeded","file":"src/core/ext/filters/deadline/deadline_filter.cc","file_line":69,"grpc_status":4}"
>
[WARN] [1603362611.335338]: Error while get launch configuration from grpc://lj-lenovo-xiaoxin-700-15isk:12321: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "Deadline Exceeded"
	debug_error_string = "{"created":"@1603362611.325307754","description":"Deadline Exceeded","file":"src/core/ext/filters/deadline/deadline_filter.cc","file_line":69,"grpc_status":4}"
>
[WARN] [1603362618.157696]: Error while get_system_diagnostics from grpc://lj-lenovo-xiaoxin-700-15isk:12321: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "Deadline Exceeded"
	debug_error_string = "{"created":"@1603362618.155295555","description":"Deadline Exceeded","file":"src/core/ext/filters/deadline/deadline_filter.cc","file_line":69,"grpc_status":4}"
>

lijian636, 2020-10-22:
Excuse me, I have studied this error is the router problem, I change a router, everything is normal,thank you very much.

closed 2020-11-11 by atiderko