fkie/multimaster_fkie #140: [Question] can the package be built with catkin_make?

skyrobos, 2020-10-22:
If not how can I install it and then use the catkin_make?
Thank you for your support

atiderko, 2020-10-22:
Yes, I built the master branch with catkin_make on Ubunu 18.04 with ROS melodic

skyrobos, 2020-10-22:
I cloned the master, run catkin_make, and got this error:
```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/roblabo/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/roblabo/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed 
```
and this is the warning that appeared before:
```
-- ~~  - fkie_multimaster (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'fkie_multimaster' contains non standard content. Use the content of the following file instead: /home/roblabo/catkin_ws/build/catkin_generated/metapackages/fkie_multimaster/CMakeLists.txt

```

I work on Ubuntu 16.04 with ROS Kinetic

atiderko, 2020-10-22:
for Ubuntu 16.04 try with branch `kinetic-daemon-devel`, please.

skyrobos, 2020-10-22:
That solved the issue. Thank you

closed 2020-10-22 by skyrobos