fkie/multimaster_fkie #141: Adjust the number of launch files shown in the launch_files tab

weixuanzhang, 2020-10-29:
Would it be possible to adjust the number of launch files shown in the launch_files tab? Consider the following example:
[image]
The most recent five launch files are shown. However, I usually have around 10 launch files (and different combinations of them) that I regularly use. Would it be possible to show more? Otherwise, I always have to manually dig into the package launch files and this is time-consuming. Thank you very much!

atiderko, 2020-10-29:
you can change the `Launch history length` parameter in the `Settings` tab
http://fkie.github.io/multimaster_fkie/chapter_usage/settings.html

weixuanzhang, 2020-10-29:
Thank you very much! I will close the issue.

closed 2020-10-29 by weixuanzhang