fkie/multimaster_fkie #142: Sync Hosts / Ignore Hosts does not appear to be working

chris-ferone, 2020-11-06:
Hi, 

I am running branch 1.1.0 on Kinectic / Ubuntu 16.04 LTS. I am trying to sync two masters on a network with as many as 8 masters at any given time. 

I started out trying to use the sync hosts parameter to specify which hosts I wanted to be discovered, but that did not seem to be working. I then set the ignore_hosts parameter to ['*'], and am I still seeing the issue. 

In the screenshot below, I am launching from one host, linux-alpha, and hoping for it to discovery only tmti2; however, you can see that it also discovers tmtvpp1. 

Note that on the tmti2 host, the multimaster setting are identical except that the sync_host parameter value is linux-alpha. On the tmtvpp1, the settings are identical as well, except the sync host parameter value is a completely different host I have even mentioned.

[image]

atiderko, 2020-11-06:
Hi,

you can not use the filter parameter to prevent hosts to be discovered. You can separate your network using `mcast_port`.
The filter interface with ignore_hosts or sync_hosts should be applied to `master_sync`.

The filer interface in master_discovery was added as experimental feature (not documented) and supports only few selected parameter. Since it uses the same class, parameter are printed out in the console it set. Officially supported parameter are listed on http://wiki.ros.org/master_discovery_fkie

chris-ferone, 2020-11-10:
Thanks so much, using the mcast_port parameter solved the problem. I did notice there is as much as a ~30 second delay before other masters are discovered. Do you think that is potentially related or should I open a separate issue for that?

atiderko, 2020-11-11:
It was a bug, already fixed in master branch, but not pushed into kinetic-daemon-devel. I've done this now.

If you can not update, you can also add `_heartbeat_hz:=0.5` parameter to `master_discovery` start.

atiderko, 2020-11-11:
please reopen if issue still exists!

closed 2020-11-11 by atiderko