fkie/multimaster_fkie #144: ModuleNotFoundError: No module named 'fkie_multimaster_msgs.grpc.file_pb2_grpc'

simchanu29, 2020-11-26:
I can't manage to launch node_manager.
Is there a specific way to run it or unlisted dependencies ?

### Context : 
OS : Ubuntu 20.04
ROS : noetic

### Step to reproduce : 
1. `sudo apt install ros-noetic-fkie-multimaster`
2. `roscore &`
3. `rosrun fkie_node_manager node_manager`

### Error

```
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/fkie_node_manager/node_manager", line 5, in <module>
    import fkie_node_manager
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager/__init__.py", line 45, in <module>
    from fkie_node_manager_daemon import host as nmdhost
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/__init__.py", line 45, in <module>
    from .server import GrpcServer
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/server.py", line 41, in <module>
    import fkie_multimaster_msgs.grpc.file_pb2_grpc as fgrpc
ModuleNotFoundError: No module named 'fkie_multimaster_msgs.grpc.file_pb2_grpc'
```

atiderko, 2020-11-27:
This is indeed a problem as the auto-generated files were not included in the Debian package.
I will try to fix it.

In the meantime you have to use the noetic-devel branch to be able to use node manager!

Thank you for reporting!

tkazik, 2021-02-10:
Thx for writing that down @simchanu29, I just ran into the same issue.

@atiderko, I was able to build the noetic-devel branch without any warnings. 
However, when trying to run the node_manager, I get the following error:
`AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'`

My setup:

- `python3-grpcio` and  `python3-grpc-tools` are installed
- `protoc --version` returns: libprotoc 3.6.1

Any ideas why it is failing? Am I missing something? Thx!

atiderko, 2021-02-11:
@tkazik : did you tried this [solution on stackoverflow.com](https://stackoverflow.com/questions/61922334/how-to-solve-attributeerror-module-google-protobuf-descriptor-has-no-attribu)?

tkazik, 2021-02-11:
Thx for the quick reply! Yes, I tried that yesterday without success.

- Using `pip install --upgrade protobuf` the version gets bumped to 3.14.0
- Now, I upgraded the official protobuf [package](https://packages.ubuntu.com/focal/protobuf-compiler) (which is at version 3.6.1 for focal) using this [method](https://grpc.io/docs/protoc-installation/#install-pre-compiled-binaries-any-os).
- Now both `pip show protobuf` and `protoc --version` show 3.14.0 and `node_manager` spawns  :tada:

The only thing I now encouter about every second time is the following error:

```console
...
[INFO] [1613065385.776784]: start node manager daemon for http://P1G3:11311/
[WARN] [1613065385.827447]: Error while list_path from grpc://p1g3:12321: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "failed to connect to all addresses"
	debug_error_string = "{"created":"@1613065385.823932252","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":4142,"referenced_errors":[{"created":"@1613065385.823930622","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":397,"grpc_status":14}]}"
...
```

However, I don't know whether that is related to protobuf or something else. Have you seen that before?

atiderko, 2021-02-11:
did you removed the build files after you updated protobuf and protoc?

Node Manager tries to connect to the node manager daemon which should be started automatically. On errors you see these warnings.

Try  `rosrun fkie_node_manager_daemon node_manager_daemon` to see if there is all ok.

tkazik, 2021-02-11:
Yes, I cleaned the build files after updating: Still the same error.
Now, running `rosrun fkie_node_manager_daemon node_manager_daemon` before `node_manager`, I observe the following:

1. `rosrun fkie_node_manager_daemon node_manager_daemon` prints out the following:

```console
...
[INFO] [1613078424.309408]: Create monitor servicer
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1222: RuntimeWarning: ignoring OSError(6, 'No such device or address') for file '/sys/class/hwmon/hwmon6/temp8_input'
  warnings.warn("ignoring %r for file %r" % (err, path),
[INFO] [1613078424.425969]: Create launch manger servicer
...
```

2. `node_manager` does not show any warnings anymore (which is good I guess).

Does this help you? Or make sense? Thx for your awesome support!

atiderko, 2021-02-12:
this looks good!

The warning only affects the sensor information, which is read out or not.

If you see launch files in launch dock of the node manager than all should be ok.

tkazik, 2021-02-12:
Ah ok...guess that is good news then :)
Yes, I see launch files in the dock (=> so I am just ignoring the [WARN] output on the console for now).
Interesting fact: If I close node_manager without shutting down the master and restart the node_manager, I don't get any warning at all. Strange...

tkazik, 2021-04-20:
A maybe related follow-up on this one: I now have troubles discovering the robot on the latest noetic-devel branch and get the following errors:
[image]
Do you have an idea what that temporary name resolution failure might be? Thx!

PS: I can ssh into the robot without issues from the terminal.

atiderko, 2021-04-21:
It looks like `paramiko` and `ssh` resolve names following different paths. `paramiko` is the library I use to handle SSH connections.

Unfortunately I cannot answer what causes `temporary failure name resolution`.

Did you try to use the IP address instead of `robot`?

sebastianfast, 2021-05-27:
For the record

We stumbled across the same problem, installing `multimaster_fkie` via `apt`. We instead manually installed the package via catkin, as @atiderko suggested, which resolved the problem for our side. We are running the multimaster inside a docker image using Ubuntu 20.04 on an Ubuntu 20.04 host.

atiderko, 2021-05-27:
@sebastianfast thank you for reporting. Currently I still did not found a solution to this problem. I don't know why the generated code will not be included into the package on Ubuntu 20.04...

Sr4l, 2021-07-22:
@atiderko my guess is the missing dependency of "python3-grpc-tools". I don't know how rosdep get the required package names, but as i can see with a fresh ubuntu20.04 and noetic the package "python3-grpc-tools" is not going to be installed and building of the node_manager fails.

jk-ethz, 2021-11-25:
> @atiderko my guess is the missing dependency of "python3-grpc-tools". I don't know how rosdep get the required package names, but as i can see with a fresh ubuntu20.04 and noetic the package "python3-grpc-tools" is not going to be installed and building of the node_manager fails.

Made no difference for me. Still not working.

Sr4l, 2021-11-25:
Have you removed node-manager (apt remove ros-noetic-fkie-multimaster) and build it in your ros working dir?

jk-ethz, 2021-11-25:
> Have you removed node-manager (apt remove ros-noetic-fkie-multimaster) and build it in your ros working dir?

No, so you mean I need that custom build and the package python3-grpc-tools?

tkazik, 2021-11-25:
Yes you need to uninstall/remove the apt version and then install the noetic version from source. So:

```sh
#  remove the official from apt: 
sudo apt install ros-noetic-fkie-multimaster

# go to your catkin workspace and clone this repo
git clone git@github.com:fkie/multimaster_fkie.git
```
Then, check out the `noetic-devel` branch and build :)

Hope that helps!

jk-ethz, 2021-11-25:
Thanks, will try!

mdlugosz-agh, 2022-07-14:
Hi, so there are any chances to correct this error. I'm install fkie*  from apt and have the same problem with node_manger - not run - error - [ModuleNotFoundError: No module named 'fkie_multimaster_msgs.grpc.file_pb2_grpc'](https://github.com/fkie/multimaster_fkie/issues/144#)

atiderko, 2022-07-18:
Hi, I was able to fix the error. However, now you have to wait for the sync for Noetic. Or you install the packages [ros-testing](http://packages.ros.org/ros-testing/ubuntu) repository

Spir0u, 2022-10-28:
I tried the same thing on Ubuntu 18.04, ros melodic. On the melodic-devel branch it works, but as I want to connect with a Noetic ROS_MASTER I tried it with the `melodic-daemon-devel` branch, where it failed.

After doing it like [above](https://github.com/fkie/multimaster_fkie/issues/144#issuecomment-777681155)
```
PB_REL="https://github.com/protocolbuffers/protobuf/releases"
curl -LO $PB_REL/download/v3.17.3/protoc-3.17.3-linux-x86_64.zip
unzip protoc-3.17.3-linux-x86_64.zip -d $HOME/.local
export PATH="$PATH:$HOME/.local/bin"
```
`pip show protobuf` has `Version: 3.17.3`, `protoc --version` returns `libprotoc 3.17.3`.

I cleaned the workspace, installed it again and I get this error:
```
~/catkin_ws/src/multimaster_fkie (melodic-daemon-devel)$ node_manager
Traceback (most recent call last):
  File "/home/geranos/catkin_ws/devel/bin/node_manager", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/geranos/catkin_ws/src/multimaster_fkie/fkie_node_manager/nodes/node_manager", line 5, in <module>
    import fkie_node_manager
  File "/home/geranos/catkin_ws/devel/lib/python2.7/dist-packages/fkie_node_manager/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 45, in <module>
  File "/home/geranos/catkin_ws/devel/lib/python2.7/dist-packages/fkie_node_manager_daemon/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 45, in <module>
  File "/home/geranos/catkin_ws/src/multimaster_fkie/fkie_node_manager_daemon/src/fkie_node_manager_daemon/server.py", line 43, in <module>
    import fkie_multimaster_msgs.grpc.file_pb2_grpc as fgrpc
  File "/home/geranos/catkin_ws/devel/lib/python2.7/dist-packages/fkie_multimaster_msgs/grpc/file_pb2_grpc.py", line 5, in <module>
    from fkie_multimaster_msgs.grpc import file_pb2 as fkie__multimaster__msgs_dot_grpc_dot_file__pb2
  File "/home/geranos/catkin_ws/devel/lib/python2.7/dist-packages/fkie_multimaster_msgs/grpc/file_pb2.py", line 21, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: 'module' object has no attribute '_internal_create_key'
```

atiderko, 2022-10-28:
The error comes from generated code... did you tried to install the version 3.14.0?

Spir0u, 2022-11-09:
I tried this too but there were still errors coming up, so I gave up and installed Ubuntu 20.04 on both.