fkie/multimaster_fkie #145: running multiple simulators with its own masters in a single computer

thezest76, 2020-11-30:
Hello,

I am trying to create two Unity simulation environments in a single PC. 

I've tried to follow the instruction in "https://answers.gazebosim.org//question/23833/how-to-start-multiple-gazebo-simulations-with-own-rosmasters/". I was able to spawn two independent simulators. Say, A and B. However, the two rosmasters are completely disconnected, That is, I am not able to see the topics in A from B.  

For example, if run a command in "rostopic pub -r 1 /test std_msgs/Int32 1" from A followed by "rostopic list" at another terminal shows:

/diagnostics
/master_discovery/changes
/master_discovery/linkstats
/rosout
/rosout_agg
/test

However, running "rostopic list" at a terminal associated with the address B shows:

/diagnostics
/master_discovery/changes
/master_discovery/linkstats
/rosout
/rosout_agg

the topic /test is missing here. 

The only difference is that the instruction is about establishing Gazebo simulators while I am launching Unity simulators. 
What bothers me is that the two masters have different _mcast_group and _mcast_port numbers, preventing sync when master_sync is executed.

Could you please let me know to sync two masters in a single PC?

Best

atiderko, 2020-11-30:
Hi,
if you change _mcast_port, you separate the networks so that several groups can exist in parallel without synchronization.

Please use the tutorial on [http://wiki.ros.org/multimaster_fkie/Tutorials](http://wiki.ros.org/multimaster_fkie/Tutorials/Setup%20a%20ROS%20master%20synchronization) to start and synchronize multiple roscore's on one host.

regards

thezest76, 2020-11-30:
Hello,

Thanks for the suggestion.
I've tried 1.1 and 1.2. But, it didn't work for me. "rostopic echo /test/topic &" command doesn't print out /test/topic message.

Instead, what I get is "WARNING: topic [/test/topic] does not appear to be published yet"
Besides, it seems that the tutorial shows wrong commands. It should be "rosrun fkie_master_sync master_sync" while the tutorial shows "rosrun master_sync_fkie master_sync", right ? 

Could please tell me if I am missing anything here ?

atiderko, 2020-11-30:
you 're right the tutorial is old ;-). It should be `fkie_master_discovery` and `fkie_master_sync`.

Just to be sure, you did the following in first terminal:
```
export ROS_MASTER_URI=http://localhost:11311
roscore --port 11311 >/dev/null 2>&1 &
rosrun fkie_master_discovery master_discovery >/dev/null 2>&1 &
rosrun fkie_master_sync master_sync >/dev/null 2>&1 &
rostopic pub /test/topic std_msgs/String 'Hello World' -r 1
```
and this in the second terminal:
```
export ROS_MASTER_URI=http://localhost:11312
roscore --port 11312 >/dev/null 2>&1 &
rosrun fkie_master_discovery master_discovery >/dev/null 2>&1 &
rosrun fkie_master_sync master_sync >/dev/null 2>&1 &
rostopic echo /test/topic
```

After both are started, which result you get calling `rosservice call /master_discovery/list_masters`

thezest76, 2020-11-30:
Well

I cannot even proceed to the last command of the 2nd terminal.
When I try to do "rosrun fkie_master_discovery master_discovery" at localhost:11312.
I get the error message below:

`
[FATAL] [1606735760.029156]: Unable to bind multicast to interface: 226.0.0.0, check that it exists: [Errno 98] Address already in use
[ERROR] [1606735760.030495]: Traceback (most recent call last):
  File "<string>", line 113, in main
  File "/home/hankm/catkin_ws/src/multimaster/fkie_master_discovery/src/fkie_master_discovery/master_discovery.py", line 602, in __init__
    self._init_socket(True)
  File "/home/hankm/catkin_ws/src/multimaster/fkie_master_discovery/src/fkie_master_discovery/master_discovery.py", line 634, in _init_socket
    self.socket = DiscoverSocket(self.mcast_port, self.mcast_group, send_mcast=self._send_mcast, listen_mcast=self._listen_mcast)
  File "/home/hankm/catkin_ws/src/multimaster/fkie_master_discovery/src/fkie_master_discovery/udp.py", line 177, in __init__
    self.bind((to_group, port))
OSError: [Errno 98] Address already in use

Error while start master_discovery: [Errno 98] Address already in use
`

thezest76, 2020-11-30:
Never mind.

It is working now. I was using a source compiled version. The one has noetic-v1.2.4 tag.
It is working after I reinstalled fkie_multimaster pkg using sudo apt-get command.

Thank you so much for your help.

atiderko, 2020-11-30:
I'm glad it works now, although it should also work with the source version.

closed 2020-11-30 by atiderko