fkie/multimaster_fkie #146: sync_hosts with ports instead of hostname

gbcarlos, 2020-12-05:
Hi,

I am running 5 masters on one pc under different ports. Now I am trying to sync all of the topics from 1 port to the other 4, ignoring all other topics. Is there a possibility to forward the ros master uri instead of the host name? As it is running on one pc the host name should be the same for all masters, correct me if I am wrong. I am working with ROS Melodic.
Any help is appreciated!

atiderko, 2020-12-08:
Hi,

if I understand you correct, you try to sync to one master and ignore all other. The `master_sync` has a parameter `sync_hosts`.  The values are names of masters. By default the host name is used. On the same host and different port, the port number will be append, e.g. `myhost_11312`.
You can also change the name while start `master_discovery`, e.g.
```
rosrun fkie_master_discovery master_discovery _name:=abc
```
And then start sync on other hosts:
```
rosrun fkie_master_sync master_sync _sync_hosts:=[abc]
```

gbcarlos, 2020-12-08:
Thanks! Thats exactly what I was looking for.

closed 2020-12-08 by gbcarlos