fkie/multimaster_fkie #148: master_discovery memory leak

amilcarlucas, 2020-12-10:
I'm using the latest master version compiled on NVidia Jetson TX2, TX2i and NX.
All three boards run on Ubuntu 18.04 LTS.

The longer the master_discovery service runs the more memory it needs.
I suspect it has a memory leak!

atiderko, 2020-12-16:
I run some tests with five ROS-Master's (five `master_discovery`, five `master_sync` and one `node_manager`) on Ubuntu 18.04.

I record the memory usage with `psrecord`.

With no changes on ROS-Master the memory usage increases slowly (almost an hour) until an upper limit (60MB)
[image]

With a ROS-Node stop/start every 5 seconds the limit of 60 MB is reached faster and stay at this level.
[image]

I don't know exactly what the reason for the increase is at the beginning and whether it can be fixed at all.

closed 2021-05-26 by atiderko

amilcarlucas, 2022-07-06:
I now upgraded to Ubuntu 20.04 LTS and ROS noetic
After a couple of days it increases from 40MiB to 140MiB

amilcarlucas, 2022-08-24:
Any progress on this?
Please re-open it.