fkie/multimaster_fkie #149: Thread.isAlive() has been removed in Python 3.9 in favor for is_alive().

ProgenitorX, 2021-01-14:
https://github.com/fkie/multimaster_fkie/blob/52f3250f5b5e87ae988260d67346a32d2ff92382/fkie_master_discovery/src/fkie_master_discovery/master_monitor.py#L596

This threw a deprecated warning in Python 3.8, but now has been removed entirely in Python 3.9. I manually changed it to is_alive() and no issues so far. Should work with any Python 3.5+ versions since it's been there since, with the same functionality as isAlive().

atiderko, 2021-01-15:
Thank you for reporting!

closed 2021-01-15 by atiderko

amilcarlucas, 2021-02-18:
Does this mean that it now requires Python >= 3.5 ?

atiderko, 2021-02-18:
> Does this mean that it now requires Python >= 3.5 ?

You can still use multimaster with Python 2.7