fkie/multimaster_fkie #152: Cannot ping network computers neither discover masters

msualeh, 2021-04-03:
I am followig the procedure detailed in your technical report. My setup has eithernet based local ros networks (single ros master each) , and a computer from each local network needs to communicate over wifi to a common ros master. I am stuck with routing step "route add -net <local network> netmask 255.255.255.0 gw <common network gateway>", ping fails even among the local computers.  I am using Ros melodic with ubuntu 18.04. Can you please suggest possible reasons for this behaviour?

atiderko, 2021-04-04:
If local routing fails, the routing table appears to be faulty. But it's hard to tell what's going on without more information.

Do you try to setup the network like shown in Figure 10 of the report?

msualeh, 2021-04-05:
Hello, thanks for the response. Following are the details of my system the best I can describe.

[image]

**On 192.168.1.111 z**
export ROS_MASTER_URI=http:// 192.168.1.110:11311
sudo route add default gw 192.168.1.110 dev eth0
sudo sh -c "echo 0 >/proc/sys/net/ipv4/icmp_echo_ignore_broadcasts"
sudo service procps restart
/etc/hosts
127.0.0.1 	localhost
127.0.1.1 	z
192.168.1.111 	z
192.168.1.110 	a
192.168.0.121 	lenovo

z@z:~$ ip r
default via 192.168.1.110 dev eth0 
default via 192.168.1.1 dev eth0 proto static metric 20100 
169.254.0.0/16 dev eth0 scope link metric 1000 
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.111 metric 100

z@z:~$ netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.110   0.0.0.0         UG        0 0          0 eth0
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 eth0
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth0
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0

**On 192.168.1.110 a**
export ROS_MASTER_URI=http:// 192.168.1.110:11311
sudo sh -c "echo 1 >/proc/sys/net/ipv4/ip_forward"
sudo service procps restart
sudo route add -net 192.168.1.0 netmask 255.255.255.0 gw 192.168.0.104
sudo sh -c "echo 0 >/proc/sys/net/ipv4/icmp_echo_ignore_broadcasts"
sudo service procps restart
/etc/hosts
127.0.0.1 	localhost
127.0.1.1 	a
192.168.1.110 	a
192.168.1.111 	z
192.168.0.121 	lenovo

a@a:~$ ip r
default via 192.168.1.1 dev eth0 proto static metric 20100 
default via 255.255.255.0 dev wlan0 proto static metric 20600 
169.254.0.0/16 dev wlan0 scope link metric 1000 
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.104 metric 600 
192.168.1.0/24 via 192.168.0.104 dev wlan0 scope link 
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.110 metric 100 
255.255.255.0 dev wlan0 proto static scope link metric 20600

a@a:~$ netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 eth0
0.0.0.0         255.255.255.0   0.0.0.0         UG        0 0          0 wlan0
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 wlan0
192.168.0.0     0.0.0.0         255.255.255.0   U         0 0          0 wlan0
192.168.1.0     192.168.0.104   255.255.255.0   UG        0 0          0 wlan0
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0
255.255.255.0   0.0.0.0         255.255.255.255 UH        0 0          0 wlan0

**On 192.168.0.121 lenovo**
export ROS_MASTER_URI=http:// 192.168.0.121:11311
sudo sh -c "echo 0 >/proc/sys/net/ipv4/icmp_echo_ignore_broadcasts"
sudo service procps restart
/etc/hosts
127.0.0.1 	localhost
127.0.1.1 	lenovo
192.168.0.121 	lenovo
192.168.1.110 	a
192.168.1.111 	z

Issue is that the moment I define route on default gateway. even local computers fail to ping

atiderko, 2021-04-08:
sorry for delayed answer!

I think the entry `sudo route add -net 192.168.1.0 netmask 255.255.255.0 gw 192.168.0.104` cause problems. Therefore, all packets to 192.168.1.111 are sent on the WLAN.

Instead you can try to enable the IP forwarding on 192.168.1.110, see e.g. [https://unix.stackexchange.com/questions/449468/how-to-route-between-interfaces](https://unix.stackexchange.com/questions/449468/how-to-route-between-interfaces)

msualeh, 2021-04-09:
Oh, I thank and appreciate that you responded.