fkie/multimaster_fkie #153: One master and multiple master

miku54, 2021-05-17:
Hello, which is better to use this feature package to synchronize specific topics or to subscribe to host topics directly under the same master?

atiderko, 2021-05-17:
Hi, it depends on your scenario, in a static scenario it is easier to just change the ROS_MASTER_URI than to install the multimaster package. However, if you have several robots and control panels, which one should host the roscore? If this changes, the environment variable must be adapted on all hosts. If the connection to the roscore is lost, no new nodes can be started.
Then it makes sense to use the multimaster.

And once you have installed multimaster, I would always use it and not change the environment variables. Especially if you use the Node Manager to launch the ROS system.

miku54, 2021-05-18:
Thank you for your reply. I get it.

closed 2021-05-26 by atiderko