fkie/multimaster_fkie #154: Can two computers run two Gazebo and two RVIZ after running multimaster_fkie?

15993217652, 2021-05-20:
Can two computers run two Gazebo and two RVIZ after running multimaster_fkie?  
Why is it that when one computer runs Gazebo and another runs Gazebo, it is killed

atiderko, 2021-05-20:
Multimaster synchronizes several `roscore's` without changing the names of the nodes and topics. The advantage is that the launch files for a single system and multi-core system do not have to be changed.
However, if two nodes with the same name are started in the system, these nodes will be terminated.
This is ROS behaviour!

To prevent this, the same nodes must be started in different namespaces.
Or you add these nodes to the `master_sync` ignore list.

15993217652, 2021-05-20:
Thank you very much for your reply. I will try again

15993217652, 2021-05-21:
Now, I can run two Gazebo on two computers, but I want to subscribe to topics for neglected nodes. Is that allowed

atiderko, 2021-05-26:
if you put the gazebo in a different namespace, you can subscribe to topics. If they're on the ignore list, then not.

15993217652, 2021-05-26:
Thank you very much for your reply. Now I have understood everything

atiderko, 2021-05-26:
you are welcome! I am sorry for the late reply

closed 2021-05-26 by atiderko