fkie/multimaster_fkie #155: Cant connect Raspberry pi and dekstop computer

tinnx, 2021-05-26:
I am trying to connect my raspberry pi and computer that are on the same network. I have on both OS Ubuntu 20.04 and ROS Noetic installed. i followed  pdf (https://digital.csic.es/bitstream/10261/133333/1/ROS-systems.pdf) with instruction but this two masters cant see each other. I can ping machines and when i start nodes master_discovery and master_sync on both machines there is no error but they don't detect each other.

[image]

[image]

[image]

[image]

I have set on both machines this instructions:

sudo sh -c "echo 0 >/proc/sys/net/ipv4/icmp_echo_ignore_broadcasts"
sudo sh -c "echo 1 >/proc/sys/net/ipv4/ip_forward"

And i had configure with correct IP addresses file _/etc/hosts_ on both of the machines.

atiderko, 2021-05-26:
Please, try to run the master_discovery with
`rosrun fkie_master_discovery master_discovery _mcast_group:=224.0.0.251 _robot_hosts:=[ubuntu,ros]`

If they can discover each other the multicast doesn't work properly.

It looks like you've set the interface variable too. Therefore the master_discovery binds to the interface with the IP 192.168.46.100. You have to make sure that the multicast messages are also forwarded on this interface.

If that was not the intention, it would be best to delete all variables from the ROS parameter server or restart roscore.

tinnx, 2021-05-26:
I restarted roscore on both machines and after that run master discovery the why you asked. On computer he gave me this message: 

[image]

And on the raspberry it gave me this message:

[image]

tinnx, 2021-05-26:
After i had restarted roscore on both machines and run command you gave me it worked and connection have been established. Problem that I have now is temporary failure in name resolution:

[image]

[image]

atiderko, 2021-05-26:
sorry, my fault! Use `_robot_hosts:=[ubuntu,pc]` instead of `_robot_hosts:=[ubuntu,ros]`

But you should try to get it runs without `_robot_hosts`, since for each entry a unicast message is send to discover the remote host.

tinnx, 2021-05-29:
Thank you for the help now everthing is working. After you told to try with unicast I figure it out that problem was on raspberry Pi. It didn't had package python-is-python3 so node master_discovery on raspberry actually didn't worked and didn't show any error on start.

closed 2021-05-29 by tinnx