fkie/multimaster_fkie #156: rosrun: command not found

lijian636, 2021-05-31:
When I want to start the daemon node of the remote computer on the local machine through the prompt of the interface, the following error message will occur, I don’t know if it can be used？
Traceback (most recent call last):
  File "/home/lj/catkin_jiedian2/src/multimaster/fkie_node_manager/src/fkie_node_manager/progress_queue.py", line 301, in run
    self._target(*self._args)
  File "/home/lj/catkin_jiedian2/src/multimaster/fkie_node_manager/src/fkie_node_manager/start_handler.py", line 158, in runNodeWithoutConfig
    raise StartException(''.join(['The host "', host, '" reports:\n', error]))
StartException: The host "eslam-board" reports:
bash: rosrun: command not found

atiderko, 2021-05-31:
To run a node on remote host, an SSH connection will be established without setting any environment variables. The environment on remote host is configured through its .bashrc. Put the `source /.../ros/.../setup.bash` before `[ -z "$PS1" ] && return` in .bashrc of the remote host.

atiderko, 2021-09-16:
I assume it was solved, please reopen if not!

closed 2021-09-16 by atiderko

antoan, 2023-03-01:
I get a similar problem on noetic-devel with zsh.

I've sourced ros noetic at the very top of my .zshrc config. 

Double clicking on the remote host's nodes to access its screen yeilds: _zsh:1: command not found: rosrun_