fkie/multimaster_fkie #158: Not ignoring topics

caiofis, 2021-07-13:
First of all thank you for the amazing repo, the documentation provided by the technical report is really amazing.
I am testing the multimaster architecture to deploy in a system with very limited bandwidth. I am not sure if it is the expected behavior but once I add a topic to the `ignore_topics` param it don't show up when I run `rostopic list`, but if I try to echo this topic it seems to be synced. I expected to see this behavior only if the flag `sync_topics_on_demand` is set to true. Am I missing something?

atiderko, 2021-07-13:
Thanks for the praise!
Did you run only one `master_sync`? If not, did you added this topic to all `master_sync` nodes?
Otherwise I need the output from the master_sync node.

atiderko, 2021-07-14:
Due to the structure of the ROS communication, the system structure you have described is not completely possible. Topics mit gleichem Name, die auf allen Roboter verwendet werden, werden auch durch einen ROS-Core miteinander verbunden, e.g. `/tf`.
If the robots do not have any common topics, e.g. because they are separated by namespaces, there are various options:
1. Prevent the robots to detect each other. Start `master_discovery` on robots with `_listen_mcast:=False`
   `rosrun fkie_master_discovery master_discovery  _listen_mcast:=False`
2. Configure `master_sync` on robots to ignore other robot hosts using `ignore_hosts` parameter
3. Configure `master_sync` on robots to sync only to control station using `sync_hosts` parameter

I hope these solutions are suitable to you

caiofis, 2021-07-16:
I think that this third option will solve my problem. Thank you!

closed 2021-07-16 by caiofis