fkie/multimaster_fkie #160: [noetic] time delay and TypeError

tkazik, 2021-07-18:
I am trying to set up a similar scenario as in this [issue](https://github.com/fkie/multimaster_fkie/issues/155) (laptop & rpi) and till section 2.1 of [this](http://www.huyaoyu.com/technical/2018/08/27/multimaster-ros-configuration-and-multimaster-fkie.html) guide, everything worked without error.
However, when firing up node_manager, I see the following issues:

[image]
* There seems to be a time delay of about 10s
* node_manager complains about a `TypeError: expected str instance, bytes found`

Any hint what I did wrong? Thx a lot

PS: On ubuntu 20.04 with the noetic

atiderko, 2021-07-18:
it seems to be an error in node_manager while displaying error. I will fix it...
But the error is printed to the terminal:  "bash: rosrun: command not found"

Did you put `source /opt/ros/noetic/setup.bash` before ` -z "$PS1" ] && return` in `.bashrc` on rpi?

Time delay is nothing that multimaster can fix. You have to synchronize your hosts to avoid problems with ROS.
In the meantime you can use the node_manager to set the time once, see [http://fkie.github.io/multimaster_fkie/chapter_usage/host_description_panel.html](http://fkie.github.io/multimaster_fkie/chapter_usage/host_description_panel.html)

tkazik, 2021-07-19:
Hm, I don't see a line with `-z "$PS1" ] && return` in my `.bashrc`, but just added the line `source /opt/ros/noetic/setup.bash` at the beginning of `.bashrc` (I also recall that not having that line at the beginning might cause issues).

Now, the errors I get in the log window changed slightly:

```
[WARN] 19.07.2021 07:46:07.626673 (/node_manager:main_window.py:MainWindow._throttle_nmd_errrors:2370): Error while list_path from grpc://p1g3:12321: 
[WARN] 19.07.2021 07:46:07.626673 (/node_manager:update_thread.py:UpdateThread.run:127): Older master_discovery on http://rpi:11311/ detected. It does not support getUser!
[WARN] 19.07.2021 07:46:08.626673 (/node_manager:main_window.py:MainWindow._throttle_nmd_errrors:2370): Error while get_version from grpc://p1g3:12321: 
```

atiderko, 2021-07-19:
Now the `master_discovery` on rpi was started, but it is an older version, which does not have implemented `getUser` method.

About other errors: did you launched `node_manager_daemon`? If you select `rpi` on the left the `node_manager` ask for lauch the daemon. If you have launched the daemon, run it properly or crash it?

tkazik, 2021-07-21:
Thx for the fix! Unfortunately, I still get some warnings, even though starting `node_manager_daemon` explicitly, see:
[image]

atiderko, 2021-07-22:
Hi @tkazik,

for noetic the debian packets are broken, see issue #144. You have to install multimaster_fkie from github.

tkazik, 2021-07-22:
Sorry my bad, I still had the debian packets on the RPi. Thx for your support!

closed 2021-07-22 by tkazik