fkie/multimaster_fkie #162: gazebo_ros_control plugin is waiting for model URDF in parameter [/jackal/robot_description] on the ROS param server

rahulelex8, 2021-07-29:
Hi,

I'm using ROS melodic. multimaster_fkie.

getting "gazebo_ros_control plugin is waiting for model URDF in parameter [/jackal/robot_description] on the ROS param server."

Also suggest me how can i pass this URDF in parameter on ROS param server?

-------------------------These are the logs from docker terminal------------------------

Finished loading Gazebo ROS API Plugin. waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting... Finished loading Gazebo ROS API Plugin. waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting... waitForService: Service [/gazebo/set_physics_properties] is now available. Physics dynamic reconfigure ready. Camera Plugin: Using the 'robotNamespace' param: 'jackal/front_stereo/' Camera Plugin: Using the 'robotNamespace' param: 'jackal/front_stereo/' Camera Plugin (ns = jackal/front_stereo/) <tf_prefix_>, set to "jackal/front_stereo" Camera Plugin (ns = jackal/front_stereo/) <tf_prefix_>, set to "jackal/front_stereo" Loading gazebo_ros_control plugin Starting gazebo_ros_control plugin in namespace: jackal gazebo_ros_control plugin is waiting for model URDF in parameter [/jackal/robot_description] on the ROS param server.

atiderko, 2021-07-30:
Hi,

where should the URDF come from? Usually this is loaded by the launch file.

If you have loaded it on another roscore and want to synchronize the parameter, you can use `param_sync.py` from the `fkie_master_sync` package.

rahulelex8, 2021-07-30:
Hi atiderko,

first of all thanks for commenting.

I am using two docker. One side I'm running gazebo and on the other side i'm running jackal robot.
Also URDF file is called from launch file from jackal robot side container.
Two roscore are running one on gazebo container side and one on robot container side.

Please explain more how can i use param_sync.py to sync robot description.

atiderko, 2021-07-30:
On the machine where you need remote parameter run:
`rosrun fkie_master_sync param_sync.py`

It is a simple script which listen to changes published by `master_discovery` and get parameter from remote `roscore`.

rahulelex8, 2021-07-30:
I'm using ros-melodic.

both sides fkie_multimaster is running, So both side master_discovery and master_sync are running.
Also i'm able to publish and echo topic from one to another.

But I'm not able to get and set parameter in between both master.
I did 'rosparam set my_param 88" on one side, So i'm able to get and list this parameter this side only.

When I'm trying to list "rosparam list" on other side, this parameter is not displayed, So i'm not able to get this parameter.

Please let me know How can i sync parameter in between fkie_multimaster.

atiderko, 2021-07-30:
The parameter synchronization is not official part of the multimaster.
The parameter synchronization is very simple and not efficient. The changes by `rosparam set` will not be detected. Only changes on nodes or topic can trigger the parameter synchronization.

What is the output of the `param_sync.py`. Was the synchronization to the other host performed?

rahulelex8, 2021-07-30:
[image]

atiderko, 2021-07-30:
looks good to me, but perhaps something was changed in MasterProxy and should be fixed, but I have currently no time for debugging, see issue #66

atiderko, 2021-09-16:
is the problem still existing?

atiderko, 2022-02-13:
please reopen if the problem persists!

closed 2022-02-13 by atiderko