fkie/multimaster_fkie #164: Running on very limited hardware

caiofis, 2021-08-19:
I am running the multimaster on a setup with very (I mean VERY) limited hardware resources, when I run the multimaster_fkie my CPU usage sky rockets and it can't even sync the desired topic, let alone do the rest of the tasks it is suppose to do. I already reduced the synced topics to the bare minimum, but the CPU usage still around 100% when the master_discovery and sync are running. Is there any way to reduce the computational burden of this package?

atiderko, 2021-08-20:
Hi, you can try to reduce the rate how often the master_discovery check the roscore for changes, e.g.: 
`rosrun fkie_master_discovery master_discovery _rosmaster_hz:=0.1`

But if on other host a lot of changes happen (new nodes or topics), the `master_sync` will every time try to sync and will still use a lot of CPU.

caiofis, 2021-08-23:
It happens as you said, the average CPU load dropped a lot, with peaks and it actually syncs. But one behavior that I didn't expected was that it stops syncing the topisc from a particular node when I started to sync the service from a different one. 
I listed the desired nodes in the sync_nodes parameter. I tried listing the node that provides this service or adding it to the sync_services list. Anyway the behavior is the same. When it starts syncing the service it throws a timeout and after that stops syncing this particular node. It shows this msg on the terminal repeatedly after the timeout:
EXCEPTION: unknown node [/controlServer]

atiderko, 2021-08-23:
can you comment out this line [sync_thread.py#L412](https://github.com/fkie/multimaster_fkie/blob/master/fkie_master_sync/src/fkie_master_sync/sync_thread.py#L412) and try again?

caiofis, 2021-08-24:
Compiling the package from source on all my devices fixed the timeout issue. But I still facing configuration ones.
All my devices are connected to the same network, devices A and B are running multimaster_fkie and can discover each other. I can see the desired topic synced between both of them.
But I also have a device C connected to this network. It is set to use the rosmaster of device B, so they are synced by ROS itself, not by multimaster_fkie. I want to sync a topic from device C to device A, but device A only syncs nodes that are running on device B. 
Is there a way to do it without running multimaster_fkie on device C?

atiderko, 2021-08-24:
you can try the `_sync_remote_nodes:=true` for `master_sync`

caiofis, 2021-08-30:
That was what I need! Thank you so much!

closed 2021-08-30 by caiofis