fkie/multimaster_fkie #166: Multimaster connection through AWS VPN

RafaelRey, 2021-09-15:
Hi, 

I've tested the package in a local network and it works like a charm. Now I want to connect two ROS Masters in two different geographical locations, using a VPN Server (AWS VPN for example). 

I have two machines connected to the VPN, I can do ping and ssh between them so they can see each other, multicasting and ip forwarding are enabled in both machines. However I don't have ping to 224.0.0.0 (Maybe the problem is there)

Has anyone tried it or some similar setup like Peer-to-Peer (WireGuard) ? If so, which settings I need to take care of?

atiderko, 2021-09-16:
Hi, thanks for using and for the praise!

we have already tried it out successfully with zerotier vpn.
Perhaps you need to set the multicast route to VPN interface, e.g.: sudo route add -net 224.0.0.0 netmask 224.0.0.0 vpn0

Otherwise you can try without multicast by setting the `~robot_hosts` parameter of the `master_discovery`.

RafaelRey, 2021-09-16:
Thank you for your fast reply! We finally used the package without the multicast feature, as it's enough for us and it worked like a charm, both with Peer-to-Peer WireGuard VPN and Amazon AWS VPN :) Thank you very much

atiderko, 2021-09-16:
you are welcome!

closed 2021-09-16 by atiderko