fkie/multimaster_fkie #167: Sync a specific topic of a ignored node

RicardoRagel, 2021-09-17:
Hi! First let me congratulate you for this great module. 

My issue is regarding the priority of the `ignore` and `sync` parameters. I would like to ignore completely a node except one of its topics. Let's call the node `/my_node` and the desired topic to sync `/my_node/topic_1`. Then, I think this should be the parameters configuration: 

```xml
</launch>
  (...)

  <node name="master_sync" pkg="fkie_master_sync" type="master_sync" output="screen">
    <rosparam param="ignore_nodes" subst_value="True">["/my_node"]</rosparam>
    <rosparam param="sync_topics" subst_value="True">["/my_node/topic_1"]</rosparam>
  </node> 

</launch>
```

However, the topic is not sync, which is consistent with the explanation `The ~ignore_nodes and ~ignore_topics lists are processed first`, but I would like to know if it exist a workaround to sync a topic of a ignored node already implemented or if you think it could be implemented as a enhancement (I mean, without changes the current procedure).

Thank you very much!

atiderko, 2021-09-20:
Thank you!

Your configuration is currently not possible. To do this, you would probably need new parameters so as not to change the current behavior.

Currently you can only use the `sync_topic` parameter, then all other topics will be ignored.
Or you remap the required topic so that it is not in the namespace of the node.

RicardoRagel, 2021-09-30:
Yes, it seems enough for most of the cases. Thank you very very much!

closed 2021-09-30 by RicardoRagel