fkie/multimaster_fkie #170: Automatic refresh on master_discovery

gustavojoseleite, 2021-11-01:
Hi,

I have two computers (A and B) running on each the multimaster node and working correctly. However, when one of these computers (A, for example) disconnects from the network and reconnects after some time, the multimaster node of computer A cannot recognize (or fail to recognize) the other multimaster running on B.
If I run the service to list the masters, only on computer B the two masters appear, while on A only the local master appears. After calling the update method (in A), the masters recognize each other and resume running normally.
Is it possible to run refresh automatically or have some parameter to "refresh" the masters automatically?

Thanks!

atiderko, 2021-11-02:
Hi,

can you try to run `master_discovery` with `_heartbeat_hz:=0.5`.

atiderko, 2022-02-13:
please reopen if the problem persists!