fkie/multimaster_fkie #171: Multimaster with robot having multiple computers not able to publish messages

Dave-van-der-Meer, 2021-11-17:
## Issue

In my setup, I can subscribe to topics, but I can't publish into the topics created by a remote master. I can see the topics through `rostopic list` but when I publish messages, they are not received on the other end.

## Setup

I have the following setup:

- one PC with its own ROS master
- one robot with two single-board computers
  - Raspberry Pi with its own ROS master
  - Nvidia Jetson with the ROS master exported to the Raspberry Pi

I want to connect the Raspberry Pi to my PC with the Multi-Master system, but I read in the comments of [this question](https://answers.ros.org/question/252378/multimaster_discovery-multimaster_fkie-package-not-recognizing-other-master-nodes-in-multi-roscore-system/) that the ROS_IP and ROS_HOSTNAME could create some issues.

## Questions

How can I configure my system in a way that all 3 devices can communicate together? Do I need to run its own ROS master on the Nvidia Jetson? How do I need to have the ROS_IP and ROS_HOSTNAME set in order to make it work?

atiderko, 2021-11-18:
Hi Dave,

as you have already correctly recognized, the best solution would be to start own ROS master on the Jetson as well.
However, if this is not possible, you can run the `master_sync` on the PC with the parameter `sync_remote_nodes:=True`.

In most cases you don't need to set the ROS_IP or ROS_HOSTNAME. But in some scenarios you have a host which can't resolve the hostname of your PC. In this case you can use ROS_IP to force ROS to use IP addresses instead of hostnames. So the robot can recognize your PC.  Don't forget to set the IP also in ROS_MASTER_URI!

The advantage of using the ROS_IP is that no host resolution is required for communication.
The disadvantage, however, is that ROS now only communicates via this IP. If this is e.g. a WiFi interface and the connection is lost, there is also no local communication!

Dave-van-der-Meer, 2021-11-19:
Hi,
Thanks for the confirmation. In the end, I tried to keep the setup as-is so that the Nvidia Jetson exports the ROS master to the Raspberry Pi and I could make the connection to my PC work by exporting the ROS_IP of the PC and by exporting the ROS_IP of both the Jetson and the Raspberry.

But I do not export the ROS_HOSTNAME on my PC.

closed 2021-11-19 by Dave-van-der-Meer