fkie/multimaster_fkie #172: Handling network delay

Dave-van-der-Meer, 2021-11-17:
## Question

How does the multi-master system work with large delays in the network connecting two or multiple ROS masters? I have a system where large delays of up to 5 seconds can occur. Will this influence the multi-master system?

atiderko, 2021-11-18:
The multimaster nodes are only really active during startup and changes of the ROS system (e.g. start of nodes). The exchange of system information is done via gRPC-TCP. On long delay the ROS changes will be recognized later on other hosts.

The FKIE multimaster does not intervene in the ROS data transmission.
Since most standard ROS communication is done over TCP, the delay will hit ROS with all the consequences that come with TCP due to high latency.

Dave-van-der-Meer, 2021-11-18:
Okay, I understand there will be no additional issues caused by the multi-master system. Thanks.

closed 2021-11-18 by Dave-van-der-Meer