fkie/multimaster_fkie #174: Syncing parameters

caiofis, 2022-02-10:
Hi @atiderko, once again thank you for the package. I am using multimaster for a while now and it has been great.
I ran into a problem this week trying to set a parameter from a node that is not been synced. I am using two host computers, on computer A  there are lots of nodes running and since we have a limited bandwidth I do not sync every node to computer B. But when I list the ros parameters on computer B I can only see those related with the synced nodes. 
I tried to run `rosrun master_sync_fkie param_sync.py _add_ns:=False` hoping that it would sync all the parameters, but without success. Is there a way to sync the parameters of a node without adding it sync_nodes?

atiderko, 2022-02-12:
hi @caiofis I haven't used sync param for a long time. Apparently the API of MasterProxy has changed. I updated the script. Can you test it now?

atiderko, 2022-02-14:
Hi @caiofis, the parameters are only synchronized on a change event caused by adding or removing a node, topic or service.

On parameter changes using dynamic_reconfigure no parameters synchronization will be executed. `dynamic_reconfigure` sets the parameter on the node direct.

closed 2023-05-26 by atiderko