fkie/multimaster_fkie #175: module 'reduced_nm' has no attribute 'StartHandler'

tkazik, 2022-02-10:
Hi

I am encountering an issue that I am a bit clueless about (on the `noetic-devel` branch).
I set up both machines according to [this](http://www.huyaoyu.com/technical/2018/08/27/multimaster-ros-configuration-and-multimaster-fkie.html) guide and everything of section 1 worked nicely. However, when firing up node_manager, I encounter the following issue:

```bash
[WARN] [1644535863.463285]: ERROR while start 'master_discovery': b"module 'reduced_nm' has no attribute 'StartHandler'\n"
[WARN] [1644535863.464635]: start discovering on octopi failed:
	b"module 'reduced_nm' has no attribute 'StartHandler'\n"
```
The exception is raised here:

https://github.com/fkie/multimaster_fkie/blob/11c985a163969654e7bb68bea647cae9a6a25334/fkie_node_manager/src/fkie_node_manager/start_handler.py#L181

Any hint or pointer is greatly appreciated, thx!

atiderko, 2022-02-12:
Hi @tkazik 

I mean, I've seen this error message before, but I can't remember the context...

What happens if you ssh on `octopi` and call:
`rosrun fkie_node_manager remote_nm.py --package fkie_master_discovery --node_type master_discovery --node_name /master_discovery _mcast_port:=11511 _mcast_group:=226.0.0.0 _robot_hosts:=[] _heartbeat_hz:=0.5 __name:=master_discovery`

tkazik, 2022-02-12:
Hi @atiderko

Thx for the quick follow-up. I just tried your suggestion on `octopi` through ssh and unfortunately, I get the same response: 

```bash
module 'reduced_nm' has no attribute 'StartHandler'
```

No further information before/afterward and the node terminates after this output. I don't really understand this behavior, as the module should be imported on the following line:

https://github.com/fkie/multimaster_fkie/blob/56f48c2a297017165144d05201b1bbc6fe5f336c/fkie_node_manager/src/fkie_node_manager/__init__.py#L58

Somehow, I suspect this to be related to a python2 vs python3 issue...any other ideas?

atiderko, 2022-02-13:
Hi @tkazik,
it may be that python3 is more restrictive. I changed the module 'reduced_nm'. Can you please try again now?

tkazik, 2022-02-13:
Sweet, works like a charm now!
Thank you so much for maintaining and improving this nice package...even during the weekend!

closed 2022-02-13 by tkazik