fkie/multimaster_fkie #184: Unknown Error in ros-noetic, ubuntu20.04

SunggooJung, 2023-02-24:
I got this unknown error while launching the fkie node_manager_daemon

my launch file  is like this
```
<node name="default_rgb"
              pkg="fkie_node_manager_daemon" type="node_manager_daemon"
              args="-a $(find capability_common)/launch/sensor_realsense_multi.launch">
       <param name="autostart" value="true"/>
</node>
```
and the error is below
```
[WARN] [1677202835.860103]: Start server failed: Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 741, in _parse_launch
	root = parse(filename).getElementsByTagName('launch')
  File "/usr/lib/python3.8/xml/dom/minidom.py", line 1959, in parse
	return expatbuilder.parse(file)
  File "/usr/lib/python3.8/xml/dom/expatbuilder.py", line 911, in parse
	result = builder.parseFile(fp)
  File "/usr/lib/python3.8/xml/dom/expatbuilder.py", line 207, in parseFile
	parser.Parse(buffer, 0)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 9, column 30

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/launch_config.py", line 189, in load
	loader.load(self.filename, roscfg, verbose=False, argv=self.argv)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 761, in load
	launch = self._parse_launch(filename, verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 743, in _parse_launch
	raise XmlParseException("Invalid roslaunch XML syntax: %s"%e)
roslaunch.xmlloader.XmlParseException: Invalid roslaunch XML syntax: not well-formed (invalid token): line 9, column 30

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/__init__.py", line 124, in start_server
	launch_manager.load_launch_file(sfile, autostart=True)
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/server.py", line 142, in load_launch_file
	self.launch_servicer.load_launch_file(interpret_path(path), autostart)
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/launch_servicer.py", line 160, in load_launch_file
	loaded, res_argv = launch_config.load([])
  File "/opt/ros/noetic/lib/python3/dist-packages/fkie_node_manager_daemon/launch_config.py", line 218, in load
	raise LaunchConfigException(message)
fkie_node_manager_daemon.launch_config.LaunchConfigException: Invalid roslaunch XML syntax: not well-formed (invalid token): line 9, column 30
```

any suggestions?!
Thanks!

atiderko, 2023-02-24:
Hi, something in your launch file is broken:
`Invalid roslaunch XML syntax: not well-formed (invalid token): line 9, column 30`

Perhaps an invisible character in line 9?

atiderko, 2023-05-26:
Please reopen if the problem persists.

closed 2023-05-26 by atiderko