fkie/multimaster_fkie #186: zsh command not found: rosrun

antoan, 2023-03-01:
I've sourced ros noetic at the very top of my .zshrc config. 

Double clicking on the remote host's nodes to access its screen yeilds: _zsh:1: command not found: rosrun_

_Originally posted by @antoan in https://github.com/fkie/multimaster_fkie/issues/156#issuecomment-1450112052_

atiderko, 2023-03-01:
When you start a node with the NodeManager, you can see in the screen of the node_manager_daemon which command is used.
If you copy this and run it yourself in the terminal, do you also get this error message?

antoan, 2023-03-01:
My problem seems to stem from the fact that the daemon is not started automatically on the remote host (whose nodes I started manually via ssh, not via the Node Manager GUI)

The screenshot below, the log shows Node Manager fails to connect to the daemon:

[image]

Some context: 
My setup has a single ros master, with the following hosts:

**leo** (master)
**jdam** (slave)
**ronin** (runs Node Manager GUI)

I start by launching the node editor with `roslaunch fkie_node_manager nm.launch` on **ronin** after which I start master_discovery via the GUI on the host which runs the ros master - **leo** 

Also, I am able to ssh bidirectionally between the host running Node Manager GUI, and the node to which Node Manager was trying to connect to.

Perhaps I'm missing something?

antoan, 2023-03-01:
When I launch the daemon manually (ssh) on the host **jdam** and double click on one of its nodes via Node Manager (**ronin**), I copied and executed the remote command highlighted in the log below, as you suggested.  This yields the following: 

$ rosrun fkie_node_manager remote_nm.py --tail_screen_log /d455/realsense2_camera
screen logfile not found for: /d455/realsense2_camera

(screen is installed on all my hosts. )

```
[paramiko.transport][INFO] 2023-03-01 19:28:13,616: Connected (version 2.0, client OpenSSH_8.2p1)
[paramiko.transport][INFO] 2023-03-01 19:28:13,645: Authentication (publickey) failed.
[paramiko.transport][INFO] 2023-03-01 19:28:13,667: Authentication (publickey) failed.
[paramiko.transport][INFO] 2023-03-01 19:28:13,682: Authentication (publickey) failed.
[rosout][INFO] 2023-03-01 19:28:13,689: No screen for /d455/realsense2_camera on jdam found
[paramiko.transport][INFO] 2023-03-01 19:28:23,561: Connected (version 2.0, client OpenSSH_8.2p1)
[paramiko.transport][INFO] 2023-03-01 19:28:23,588: Authentication (publickey) failed.
[paramiko.transport][INFO] 2023-03-01 19:28:23,605: Authentication (publickey) failed.
[paramiko.transport][INFO] 2023-03-01 19:28:23,623: Authentication (publickey) failed.
[paramiko.transport][INFO] 2023-03-01 19:28:27,803: Authentication (password) failed.
[paramiko.transport][INFO] 2023-03-01 19:28:40,667: Connected (version 2.0, client OpenSSH_8.2p1)
[paramiko.transport][INFO] 2023-03-01 19:28:40,707: Authentication (publickey) successful!
[rosout][INFO] 2023-03-01 19:28:40,708: REMOTE execute on tony@jdam: **rosrun fkie_node_manager remote_nm.py --tail_screen_log /d455/realsense2_camera**
[rosout][WARNING] 2023-03-01 19:28:41,132: Get loggers for /d455/realsense2_camera failed; cannot get service URI from http://jdam:11311/: [Errno 111] Connection refused
```

atiderko, 2023-03-02:
The `multimaster` is designed for use with multiple ros masters. If you set up a system with a single ros master, starting `master_discovery` makes no sense. Also, you can run only one `node_manager_daemon` per ros master. This makes it impossible to do many things with `node_manager` on hosts where `node_manager_daemon` is not running.

Now about your problem.
By double-clicking on the nodes of the remote host, `node_manager` tries to invoke its screen. If this fails, the LOG file of the node is opened (if exists). In your case the LOG file contains "__zsh:1: command not found: rosrun__". This is what you see in the `node_manager`.
Since you start the node _realsense2_camera_ manually via launch file, you have to look for the problem there. Unless the launch file contains something like __launch-prefix="screen -dmS _d455__realsense2_camera"__, node_manager will not find a screen for this node.
That's why it seems to be related to the __zsh__ after all. If you use __screen__ prefix in the launch file, you have to check if corresponding ROS_-Environment variables are available in the `screen`. Unfortunately, I haven't worked with __zsh__ yet and can't say if there are any specifics to be aware of.

antoan, 2023-03-04:
I see, thank you.  I'm guessing it won't be easy to attempt to adapt the code for my single master use case, otherwise I'd be grateful for any pointers in that direction.

Thanks again, really appreciate your thorough response.

closed 2023-03-04 by antoan