fkie/multimaster_fkie #188: Feed args to autostart launch files

LossOfHumanIdentity, 2023-03-14:
I am running multiple launch files with

```
<node pkg="fkie_node_manager_daemon"
           type="node_manager_daemon"
           name="node_manager_daemon"
           args="--autostart '$(arg _autostart_launch_files)'"/>
```

All these launch files require a robot_namespace arg to be set. How do I update the above code to feed the robot_namespace arg to all launch files?

atiderko, 2023-03-15:
Setting arguments at autostart is not supported by the daemon.

You can create another launch file that includes all launch files and sets required arguments. The new launch file does not necessarily have to be in a ROS package.

atiderko, 2023-05-26:
Please reopen if the problem persists.

closed 2023-05-26 by atiderko