fkie/multimaster_fkie #190: Both the masters which I am trying to synchronize are not showing in masters list

siddharthbhurat4, 2023-04-14:
On Master 1 - It is my laptop which is connected to the same network to which the other master (robot) is connected. I run `rosrun fkie_master_discovery master_discovery _rpc_port:=11612` on this master and then run `rosrun fkie_master_sync master_sync`

On Master 2 - It the the the toyota hsr robot which is running ros noetic in its docker, but there is now way to set an external master for this, so I decided to use this multimaster functionality. I am able to access this second master on the same laptop as Master 1. But the roscore is the one which is running on docker on the robot connected to same network. I run `rosrun fkie_master_discovery master_discovery` on this master and then run `rosrun fkie_master_sync master_sync`.

When I follow the above steps, all the commands work without error, but in the master list, I am not able to see both the master in this. Please can I know how to solve this.
Thank you.

atiderko, 2023-04-18:
`master_discovery` uses multicast to discover each other. I am not sure if these messages are forwarded to the docker container.
As a first option you can try to use the `robot_hosts` parameter. This is a list of hosts to which unicast discovery messages are to be sent. E.g. in the docker container:
```
rosrun master_discovery_fkie master_discovery _robot_hosts:=[LaptopHostname].
```

atiderko, 2023-05-26:
Please reopen if the problem persists.

closed 2023-05-26 by atiderko