fkie/multimaster_fkie #195: [WARN] can't retrieve connection information using XMLRPC from [http://192.XXX.XXX.ae:11611], socket error: timed out

mericgeren, 2023-05-31:
Hello,

I am trying to do the same what i did there: #194 in WSL version 1. I have 2 computers:

**Laptop**

-------------

- **OS:** Windows 10

- **WSL OS:** Linux Ubuntu 20.04 (I am using this for ROS and multimaster)

- **WSL Version:** 1

- **Connection Method:** Wifi

- **IP Address:** 192.XXX.XXX.ae (original ip of the laptop, same network with the other computers)

- **WSL IP Address:** Same with the laptop.

- **ROS Distro:** Noetic

- **ROS Parameters:** Set in ".bashrc" and just like following:

```
export ROS_MASTER_URI=http://laptop-windows:11311

export ROS_HOSTNAME=laptop-windows
```

- **Multimaster:** fkie_multimaster installed.

- **/etc/hosts:**

```
127.0.0.1              localhost

127.0.1.1              laptop-windows.localdomain       laptop-windows

192.XXX.XXX.ae      laptop-windows

192.XXX.XXX.dc      pc-linux
```

-------------------------------------------------------------------------------------

**PC**

------------

- **OS:** Linux Ubuntu 18.04 (non-VM)

- **VM:** None

- **Connection Method:** Ethernet

- **IP Address:** 192.XXX.XXX.dc (In the same network with the other computer)

- **ROS Distro:** Melodic

- **ROS Parameters:** Set in ".bashrc" and just like following:

```
export ROS_MASTER_URI=http://192.XXX.XXX.dc:11311

export ROS_IP=192.XXX.XXX.dc
```

- **Multimaster:** fkie_multimaster installed.

- **/etc/hosts:** 

```
127.0.0.1              localhost

127.0.0.1              pc-linux

192.XXX.XXX.ab         laptop-linux

192.XXX.XXX.ae         laptop-windows    

192.XXX.XXX.dc         pc-linux
```

I can ping each from eachother both with ips and their hostnames without any errors or duplicates. Then i did `rosrun fkie_master_discovery  master_discovery _log_level:=DEBUG _robot_hosts:=['192.XXX.XXX.XXX'] send_mcast:=False listen_mcast:=False` on both computers. This is the output in the laptop-windows:

```
[INFO] [1685535984.541164]: Check the ROS Master[Hz]: 1
[INFO] [1685535984.545838]: Heart beat [Hz]: 0.02
[INFO] [1685535984.549145]: Active request after [sec]: 60
[INFO] [1685535984.552254]: Remove after [sec]: 300
[INFO] [1685535984.556303]: Robot hosts: ['192.XXX.XXX.dc']
[INFO] [1685535984.559417]: Approx. mininum avg. network load: 2.72 bytes/s
[INFO] [1685535984.580044]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1685535984.586414]: hide_nodes: []
[INFO] [1685535984.592322]: hide_topics: []
[INFO] [1685535984.598337]: hide_services: []
[INFO] [1685535984.601438]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1685535986.800580]: Detected master discovery: http://192.XXX.XXX.ae:11611
[INFO] [1685535987.680392]: Detected master discovery: http://192.XXX.XXX.dc:11611
[INFO] [1685535987.803892]: Added master with ROS_MASTER_URI=http://192.XXX.XXX.dc:11311/
[WARN] [1685535996.918111]: can't retrieve connection information using XMLRPC from [http://192.XXX.XXX.ae:11611], socket error: timed out
[WARN] [1685536011.939376]: can't retrieve connection information using XMLRPC from [http://192.XXX.XXX.ae:11611], socket error: timed out
```
This is the output in pc-linux:

```
[INFO] [1685535984.541164]: Check the ROS Master[Hz]: 1
[INFO] [1685535984.545838]: Heart beat [Hz]: 0.02
[INFO] [1685535984.549145]: Active request after [sec]: 60
[INFO] [1685535984.552254]: Remove after [sec]: 300
[INFO] [1685535984.556303]: Robot hosts: ['192.XXX.XXX.dc']
[INFO] [1685535984.559417]: Approx. mininum avg. network load: 2.72 bytes/s
[INFO] [1685535984.580044]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1685535984.586414]: hide_nodes: []
[INFO] [1685535984.592322]: hide_topics: []
[INFO] [1685535984.598337]: hide_services: []
[INFO] [1685535984.601438]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1685535986.800580]: Detected master discovery: http://192.XXX.XXX.ae:11611
[INFO] [1685535987.680392]: Detected master discovery: http://192.XXX.XXX.dc:11611
[INFO] [1685535987.803892]: Added master with ROS_MASTER_URI=http://192.XXX.XXX.dc:11311/
[WARN] [1685535996.918111]: can't retrieve connection information using XMLRPC from [http://192.XXX.XXX.ae:11611], socket error: timed out
[WARN] [1685536011.939376]: can't retrieve connection information using XMLRPC from [http://192.XXX.XXX.ae:11611], socket error: timed out
```

Could you help me on this issue please?

Thanks in advance.

atiderko, 2023-05-31:
can you try to set the _rpc_addr:=192.XXX.XXX.ae on the laptop-windows

mericgeren, 2023-05-31:
When i try to set _rpc_addr:=192.XXX.XXX.ae with `rosrun fkie_master_discovery  master_discovery _log_level:=DEBUG _robot_hosts:=['192.XXX.XXX.dc'] send_mcast:=False listen_mcast:=False _rpc_addr:=192.XXX.XXX.ae` on inlaptop-windows, i get this warning repeating itself:

`[WARN] [1685538983.994773]: Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...`