fkie/multimaster_fkie #196: Error while start master_discovery

jfaure6, 2023-06-23:
Hello, I followed the Technical Report by Sergi Hernandez Juan and Fernando Herrero Cotarelo (Multi Master ROS Systems) and I am facing this issue : Error while start master_discovery: socket.error[19]: [Errno 19] No such device,
is multicast route set? e.g. sudo route add -net 224.0.0.0 netmask 224.0.0.0 eth0
By running the suggested command, it works fine, but for once. I need to start this command each time and it is annoying. How to fix this issue ?

atiderko, 2023-06-23:
Hello, you need to set a default gateway or route for multicast in your network settings.

jfaure6, 2023-06-28:
Thx. I saw in the document a way to set a default gateway. I tried to follow the instructions (creating a file in /etc/network/if-up.d) to add a default gateway, but I don't know where to do it, because my setup is the following :
- 4 drones with a master running on each ;
- my ground computer on which I want to synchronize the drones' masters.

atiderko, 2023-06-28:
There are many ways to set a permanent route under Linux. It depends on your operating system and chosen network configuration.
It is important to define to which network interface the multicast messages should be sent. You have to set this up on each host.