fkie/multimaster_fkie #198: ROS 2 Support?

RMichaelSwan, 2023-10-23:
I have been using this tool for quite some time on all my larger multi-pc robots. One thing that is especially useful is the node manager GUI and all the tools it provides. Is anyone working on this in some form for ROS2? Especially considering Noetic will be EOL in 2025. Obviously the multi-master part will not be relevant for ROS2, but there are many other useful things besides that.

atiderko, 2023-10-24:
Vielen Dank für das Feedback. Wir arbeiten zurzeit tatsächlich an einer Version für ROS2. Hoffentlich können die erste Version noch dieses Jahr veröffentlichen.

RMichaelSwan, 2023-10-27:
Thank you! Hope to see that soon.

Sr4l, 2024-06-17:
Is there any news regarding node_manager and ROS2? We are migrating our robots to ROS2 and I would really miss the node_manager for viewing topics, nodes and services.

We might be able to help you port the node_manager to ROS2.

RMichaelSwan, 2024-06-17:
> Is there any news regarding node_manager and ROS2? We are migrating our robots to ROS2 and I would really miss the node_manager for viewing topics, nodes and services.
> 
> We might be able to help you port the node_manager to ROS2.

It looks like the ROS2 version is called "Multi Agent Suite" now. I haven't tested, but looks promising: https://github.com/fkie/fkie-multi-agent-suite

atiderko, 2024-06-18:
As RMichaelSwan already wrote, we are working on the new version, which already supports ROS and ROS2. You can already use it, but not all ROS2 features are implemented yet.

Sr4l, 2024-06-21:
Oh, I did not catch that. Looks nice, thanks for the hard work. We will try it.