fkie/multimaster_fkie #200: All topics become invisible when ~hide_nodes="'*'" is set on all masters.

antopost, 2024-05-17:
I have two ROS masters communicating with each other.
Since both ROS masters have many nodes running with the same name and I don't want the nodes to kill each other, I would like to use ~hide_nodes to hide same-name nodes.
Since there are many same-name nodes that don't follow a particular naming convention, I tried hiding all nodes by passing ~hide_nodes="'*'" to the master_discovery.
When doing so on one ROS master everything behaves as expected: All nodes are hidden on all other masters.

**The problem arises when ~hide_nodes="'*'" on both masters is set. All topics become hidden although the masters are still being found by each other.**

Any help in solving this mystery would be appreciated.

atiderko, 2024-05-17:
No plausibility tests are performed for ~hide_nodes. If all nodes are hidden with “*”, their topics and services are of course no longer visible. This is exactly what you are describing. Unfortunately, it was integrated as a quick solution at the time. But I never got around to testing and documenting it properly.

Perhaps in your case it is better to use the parameters of [master_sync](https://wiki.ros.org/master_sync_fkie)

antopost, 2024-05-17:
Thanks for the quick response. 
Could you clarify the difference between hide_nodes and ignore_nodes

I thought hide_nodes was a way to hide only the nodes from other masters without affecting topics and message transfer.
During my tests this seemed to be the way it works. If one master_discovery was initiated with ~hide_nodes="'*'" its nodes weren't visible to the other master. Only when both masters had hide_nodes set to * did all the topics disappear.

ignore_nodes on the other hand, makes the other masters ignore all associated topics of these nodes.

Did I understand correctly?
And do you have any recommendations on how to avoid nodes with duplicate names from killing each other on synced masters?

atiderko, 2024-05-21:
_hide_nodes_ is the parameter of _master_discovery_. This means that the nodes are not visible to master_sync and node_manager.
_ignore_nodes_ is the parameter of _master_sync_. The nodes (their topics and services) from all other masters are not registered to the local ROS master.
Due to the ROS structure, the topics cannot be registered to the ROS master without nodes. For this reason, filtering the nodes also filters their topics and services.

The nodes are killed by ROS itself. In the past, only the old node was terminated; in the meantime, the new node is also terminated. 

It sounds trite now, but avoid the same names ;-)

We use namespaces.

antopost, 2024-05-21:
Ah, I understand the difference now. Thanks a bunch!

In the end I did it like you said and now use namespaces (/telemetry*) for topics _and_ nodes, then launch the master_sync with "/telemetry*" passed to sync_topics and sync_nodes. That solved the problem for me.

closed 2024-05-21 by antopost