fkie/multimaster_fkie #201: Sudden Unregistering of Remote Node (Unsync Node Register) in Multi-Machine Setup

KhairulM, 2024-09-04:
# Setup
I have 2 machines running ubuntu 20.04 and ROS Noetic on the same network:
- Machine 1 is for running a gazebo simulation (Alias: "FMS")
- Machine 2 is for spawning the robot in the simulation and running the navigation stack

I've configured the /etc/hosts of both of the machine and multicasting have been enabled (/proc/sys/net/ipv4/ip_forward set to 1 and /proc/sys/net/ipv4/icmp_echo_ignore_broadcasts set to 0)

This is my launch file for Machine 1:
```xml
<launch>
	<arg name="gui" default="false"/>
	
	<node name="master_discovery" pkg="fkie_master_discovery" type="master_discovery" output="screen">
		<rosparam param="robot_hosts">['ROBOT1', 'ROBOT2']</rosparam>
	</node>
	<include file="$(find gazebo_ros)/launch/empty_world.launch">
	    <arg name="world_name" value="$(find turtlebot3_gazebo)/worlds/turtlebot3_house.world"/>
	    <arg name="paused" value="false"/>
	    <arg name="use_sim_time" value="true"/>
	    <arg name="gui" value="$(arg gui)"/>
	    <arg name="headless" value="false"/>
	    <arg name="debug" value="false"/>
            <arg name="verbose" value="true"/>
	</include>
</launch>
```

This is my launch file for Machine 2:
```xml
<launch>
        <arg name="model" default="waffle_pi" doc="model type [burger, waffle, waffle_pi]"/>
        <arg name="robot_name" default="robot1" doc="name of the robot (must be unique and does not contain whitespaces)"/>
        <arg name="x_pos" default="-4.0"/>
        <arg name="y_pos" default="1.0"/>
        <arg name="z_pos" default="0.0"/>
        
        <node name="master_discovery" pkg="fkie_master_discovery" type="master_discovery" output="screen">
		<!-- <rosparam param="robot_hosts">['FMS']</rosparam> -->
		<rosparam param="heartbeat_hz">2</rosparam>
		<!-- <rosparam param="send_mcast">False</rosparam> -->
		<!-- <rosparam param="listen_mcast">False</rosparam> -->
	</node>
        <node name="master_sync" pkg="fkie_master_sync" type="master_sync" output="screen">
                <rosparam param="ignore_hosts"> ['ROBOT2'] </rosparam>
                <rosparam param="sync_hosts"> ['FMS'] </rosparam>
		<!-- <rosparam param="ignore_topics"> ['*theora*', '*compressedDepth*'] </rosparam> -->
		<!-- <rosparam param="resync_on_reconnect">False</rosparam> -->
        </node>

        <param name="/use_sim_time" value="true"/>
        
        <group ns="$(arg robot_name)">
                <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro' robot_name:=$(arg robot_name)" />
                <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-urdf -model $(arg robot_name)_turtlebot3 -x $(arg x_pos) -y $(arg y_pos) -z $(arg z_pos) -param robot_description -robot_namespace $(arg robot_name)" output="screen" />
                <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen">
                        <param name="publish_frequency" type="double" value="50.0"/>
                </node>
                <node name="base_link_to_laser" pkg="tf" type="static_transform_publisher" args="-0.032 0 0.1397 0 0 0 base_link base_scan 100"/>

                <!-- tranformations relay -->
                <node name="prefixed_odom_relay" pkg="tf" type="static_transform_publisher" args="0 0 0 0 0 0 odom $(arg robot_name)/odom 100"/>
                <node name="prefixed_base_footprint_relay" pkg="tf" type="static_transform_publisher" args="0 0 0 0 0 0 $(arg robot_name)/base_footprint base_footprint 100"/>
        </group>
	

	<!-- topic relay nodes -->
	<node name="$(arg robot_name)_cmd_vel_relay" pkg="topic_tools" type="relay" args="/cmd_vel /$(arg robot_name)/cmd_vel" output="screen" />
	<node name="$(arg robot_name)_imu_relay" pkg="topic_tools" type="relay" args="/$(arg robot_name)/imu /imu" output="screen" />
	<node name="$(arg robot_name)_joint_states_relay" pkg="topic_tools" type="relay" args="/$(arg robot_name)/joint_states /joint_states" output="screen"/>
	<node name="$(arg robot_name)_odom_relay" pkg="topic_tools" type="relay" args="/$(arg robot_name)/odom /odom" output="screen" />
	<node name="$(arg robot_name)_scan_relay" pkg="topic_tools" type="relay" args="/$(arg robot_name)/scan /scan" output="screen"/>
</launch>
```

I spawned a turtlebot3 model that I have edited so the topics and tf are prefixed properly and the local nodes on Machine 2 can communicate using regular topics/services name such as /odom, /scan, etc

# Problem
When I run the navigation stack (this is a custom navigation stack we develop) in Machine 2, sometimes this log appears `Unregistering /gazebo`, its pretty inconsistent to when the log appears after running the navigation stack. But when I tried to get the info of the gazebo node (`rosnode info /gazebo`) in Machine 2 I got this
[image]
The publications and services is empty on Machine 2, but on Machine 1 the gazebo node still has the correct publications and services and still publishing the correct topic

What could cause this error?

atiderko, 2024-09-04:
I don't see any configuration errors that could lead to this.

Normally a node (e.g. `/gazebo`) is defined via topics/services. Only if all are unregistered, you will not find any information about the node in roscore. That is why I am a bit surprised about the log message `Unregistering /gazebo`. I do not know where this comes from.

But the question remains why no publishers and services are displayed. This raises two questions for me:
1. is it different before the log message , i.e. directly after the start?
2. robot_hosts ROBOT1 and ROBOT2 were specified in the configuration. However, RNS3 is displayed in the screenshot. Question: are ROBOT1 and ROBOT2 correctly resolved to the IP addresses 10.10.10.4 and 10.10.10.15 on Machine1 (FMS)?  If not, it can happen that the robot sees the FSM at the beginning via the multicast communication and then not because the names on the FSM are resolved to the wrong addresses.

Perhaps the log files of `master_discovery` and `master_sync` could provide more information.

KhairulM, 2024-09-04:
Hi @atiderko thank you for the quick response, to answer your questions:
1. yes, before the `Unregistering /gazebo` log, the publications and services of /gazebo if viewed from Machine 2 are not empty, this is what it looked like 
[image]

For reference, this is the output of `rosnode info /gazebo` on FMS (Machine 1)
```
--------------------------------------------------------------------------------
Node [/gazebo]
Publications: 
 * /clock [rosgraph_msgs/Clock]
 * /gazebo/link_states [gazebo_msgs/LinkStates]
 * /gazebo/model_states [gazebo_msgs/ModelStates]
 * /gazebo/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /gazebo/parameter_updates [dynamic_reconfigure/Config]
 * /gazebo/performance_metrics [gazebo_msgs/PerformanceMetrics]
 * /rns2/camera/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns2/camera/parameter_updates [dynamic_reconfigure/Config]
 * /rns2/camera/rgb/camera_info [sensor_msgs/CameraInfo]
 * /rns2/camera/rgb/image_raw [sensor_msgs/Image]
 * /rns2/camera/rgb/image_raw/compressed [sensor_msgs/CompressedImage]
 * /rns2/camera/rgb/image_raw/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns2/camera/rgb/image_raw/compressed/parameter_updates [dynamic_reconfigure/Config]
 * /rns2/camera/rgb/image_raw/compressedDepth [sensor_msgs/CompressedImage]
 * /rns2/camera/rgb/image_raw/compressedDepth/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns2/camera/rgb/image_raw/compressedDepth/parameter_updates [dynamic_reconfigure/Config]
 * /rns2/camera/rgb/image_raw/theora [theora_image_transport/Packet]
 * /rns2/camera/rgb/image_raw/theora/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns2/camera/rgb/image_raw/theora/parameter_updates [dynamic_reconfigure/Config]
 * /rns2/imu [sensor_msgs/Imu]
 * /rns2/joint_states [sensor_msgs/JointState]
 * /rns2/odom [nav_msgs/Odometry]
 * /rns2/scan [sensor_msgs/LaserScan]
 * /rns5/camera/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns5/camera/parameter_updates [dynamic_reconfigure/Config]
 * /rns5/camera/rgb/camera_info [sensor_msgs/CameraInfo]
 * /rns5/camera/rgb/image_raw [sensor_msgs/Image]
 * /rns5/camera/rgb/image_raw/compressed [sensor_msgs/CompressedImage]
 * /rns5/camera/rgb/image_raw/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns5/camera/rgb/image_raw/compressed/parameter_updates [dynamic_reconfigure/Config]
 * /rns5/camera/rgb/image_raw/compressedDepth [sensor_msgs/CompressedImage]
 * /rns5/camera/rgb/image_raw/compressedDepth/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns5/camera/rgb/image_raw/compressedDepth/parameter_updates [dynamic_reconfigure/Config]
 * /rns5/camera/rgb/image_raw/theora [theora_image_transport/Packet]
 * /rns5/camera/rgb/image_raw/theora/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rns5/camera/rgb/image_raw/theora/parameter_updates [dynamic_reconfigure/Config]
 * /rns5/imu [sensor_msgs/Imu]
 * /rns5/joint_states [sensor_msgs/JointState]
 * /rns5/odom [nav_msgs/Odometry]
 * /rns5/scan [sensor_msgs/LaserScan]
 * /rosout [rosgraph_msgs/Log]
 * /tf [tf2_msgs/TFMessage]

Subscriptions: 
 * /clock [rosgraph_msgs/Clock]
 * /gazebo/set_link_state [unknown type]
 * /gazebo/set_model_state [unknown type]
 * /rns2/cmd_vel [unknown type]
 * /rns5/cmd_vel [unknown type]

Services: 
 * /gazebo/apply_body_wrench
 * /gazebo/apply_joint_effort
 * /gazebo/clear_body_wrenches
 * /gazebo/clear_joint_forces
 * /gazebo/delete_light
 * /gazebo/delete_model
 * /gazebo/get_joint_properties
 * /gazebo/get_light_properties
 * /gazebo/get_link_properties
 * /gazebo/get_link_state
[... 310 lines cut ...]
[rosout][INFO] 2024-09-04 19:09:35,802: Approx. mininum avg. network load: 1.36 bytes/s
[rosout][INFO] 2024-09-04 19:09:35,811: Start RPC-XML Server at ('0.0.0.0', 11611)
[rosout][INFO] 2024-09-04 19:09:35,814: hide_nodes: []
[rosout][INFO] 2024-09-04 19:09:35,816: hide_topics: []
[rosout][INFO] 2024-09-04 19:09:35,819: hide_services: []
[rosout][INFO] 2024-09-04 19:09:35,820: Subscribe to parameter `/roslaunch/uris`
[rosout][INFO] 2024-09-04 19:09:35,962: Detected master discovery: http://10.0.0.105:11611
[rosout][INFO] 2024-09-04 19:09:36,066: Added master with ROS_MASTER_URI=http://RNS5:11311/
[rospy.internal][INFO] 2024-09-04 19:09:36,148: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2024-09-04 19:09:36,821: topic[/master_discovery/changes] adding connection to [/master_sync], count 0
[rosout][INFO] 2024-09-04 19:10:07,428: Detected master discovery: http://10.10.100.23:11611
[rosout][INFO] 2024-09-04 19:10:07,539: Added master with ROS_MASTER_URI=http://FMS:11311/
[rospy.internal][INFO] 2024-09-04 19:10:08,951: topic[/clock] adding connection to [http://FMS:36921/], count 0
[rosout][WARNING] 2024-09-04 19:10:25,218: MasterMonitor[/velocity_smoother]: can't update contact information. ROS master responds with: unknown node [/velocity_smoother]
[rospy.core][INFO] 2024-09-04 19:16:37,566: signal_shutdown [signal-2]
[rosout][INFO] 2024-09-04 19:16:37,567: Unsubscribe from parameter `/roslaunch/uris`
[rosout][INFO] 2024-09-04 19:16:37,568: shutdown own RPC server
[rospy.internal][INFO] 2024-09-04 19:16:37,615: topic[/master_discovery/changes] removing connection to /master_sync
[rospy.internal][INFO] 2024-09-04 19:16:38,173: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2024-09-04 19:16:38,174: topic[/clock] removing connection to http://FMS:36921/
[rospy.impl.masterslave][INFO] 2024-09-04 19:16:38,175: signal-2
[rospy.core][INFO] 2024-09-04 19:16:38,964: signal_shutdown [atexit]
```

After I stop the navigation stack, but master_sync still up, the publications and services is still empty

KhairulM, 2024-09-04:
And for additional information, the `Unregistering /gazebo` log was only printed in our navigation stack log, but not on the master_discovery or master_sync log

atiderko, 2024-09-04:
Two things that could be checked:
1. The `master_discovery` log on the FMS says that the rns2 is no longer reporting and is therefore removed from the list.
```
[rosout][INFO] 2024-09-04 19:08:27,994: Set host to offline: rns2
[rosout][INFO] 2024-09-04 19:12:23,087: Remove master discovery: http://10.0.0.102:11611
```
  This may also be the case on rns2. Maybe you can add this parameter to all master_discovery: 
    `<rosparam param="heartbeat_hz">0.5</rosparam>`
  And also comment in the `robot_hosts` parameter again. This means that everyone proactively sends heartbeats as a unicast message. If the problem is caused by multicast, this would solve the problem.

2. The nodes register with master node, which can be reached via localhost.
```
[rospy.registration][INFO] 2024-09-04 19:09:35,676: Registering with master node http://localhost:11311
```
The ROS wiki notes that localhost in combination with remote nodes can lead to problems [(https://wiki.ros.org/ROS/EnvironmentVariables#ROS_MASTER_URI](https://wiki.ros.org/ROS/EnvironmentVariables#ROS_MASTER_URI)
Normally ROS resolves localhost to hostname, but this does not seem to be the case for you.
Although I don't think it will solve the original problem, it may lead to problems elsewhere that you don't have now.

KhairulM, 2024-09-06:
I've added the `heartbeat_hz` param to all master discovery nodes and the `robot_hosts` param to all robot's master discovery, but I still get the same log `Unregistering /gazebo` when I run my navigation stack. When I check the /gazebo node info, the publications and services are still empty
[image]

KhairulM, 2024-09-06:
After discussing with our team, this log was not printed by our code, so I guess it's from the ROS system itself

atiderko, 2024-09-09:
Have you tried to start gazebo and your navigation stack in the same launch file?

Otherwise you could subscribe to the topics `/master_discovery/linkstats` and `/master_discovery/changes` on the `RNS3` and see if you can find any hints there.
I assume that all docker containers are running on the same PC or are connected to each other via ethernet, then the quality of `/master_discovery/linkstats` should be 100. At `/master_discovery/changes` and also the service `/master_discovery/list_masters` all masters should be online.

You can also see if the same thing happens with other nodes. Start a publisher on the FSM, e.g. `rostopic pub -r 1 /test std_msgs/Bool “{data: true}”` and on the RNS3 the subscriber `rostopic echo /test`.
If this also stops working after a certain time, something is wrong with the communication between FSM and RNS3.